OPCFoundation/UA-.NETStandard #2969: Getting NodeState for i=68 (PropertyType) returns an object of type BaseDataVariableTypeState

wbtoms, 2025-01-28:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When getting the NodeState corresponding to the node PropertyType (i=68) using the method FindNodeInAddressSpace of class CustomNodeManager2, the NodeState object returned is BaseDataVariableTypeState instead of a PropertyTypeState.

This seems to be related to the loading of predefined node from the binary file Opc.Ua.Stack.Generated.Opc.Ua.PredefinedNodes.uanodes in the DiagnosticsNodeManager, see attached screenshot.

[image]

### Expected Behavior

Get PropertyTypeState instead of BaseDataVariableTypeState 

### Steps To Reproduce

Use a reference server and just call FindNodeInAddressSpace on the node manager using the node ID i=68

### Environment

```markdown
- OS: Windows 11
- Environment: VS 2022
- Runtime: .NET9.0
- Nuget Version:
- Component:1.5.374.158 but same on master
- Server: ReferenceServer or anyother server
```

mrsuciu, 2025-03-06:
@wbtoms Yes, you are right in pointing this non compliant behavior but on the other hand it is problematic to change it at this point in time since this would bring breaking changes to users of the stack that base their code on the expectance of the type being BaseDataVariableTypeState.

wbtoms, 2025-03-09:
Yes breaking changes involve some work during integration, but to my opinion, breaking changes are part of the normal life cycle of a software, it is not a bad thing espacially when it is to fix design issues or bugs. To me, it is worse to leave an issue in the software that requires tricks or introduce confusion that makes maintenance harder over years or even decades for industrial software with turnover in teams, than making a breaking change that will makes maintainability better once integrated. This is just my opinion.

mrsuciu, 2025-05-09:
Since such breaking changes are usually difficult to be traced to their source the change is not an option.

romanett, 2026-02-14:
will be fixed with 1.6 release

closed 2026-02-14 by romanett