OPCFoundation/UA-.NETStandard #2979: NETStandard doesn't pass UACTT DoS attack test

Siyuan-Xu-Wapice, 2025-02-03:
### Type of issue

- [x] Bug
- [x] Enhancement
- [x] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The NETStandard Stack doesn't comply with UCATT Security Test:
* Dos attempt by consuming Channels
* Dos attack #2; consume SecureChannels, but CreateSessions in some.

Configuration:
ReferenceServer:
MaxChannel: 76
MaxSession: 75

When the stack doesn't behave according to the OPCUA Specification, all the servers created by the stack would suffer from the same security exploit.

[image]

[image]

### Expected Behavior

According to OPC UA Specification [Core/Part4/v105/docs/5.6.2](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.6.2): 
_"To protect against misbehaving Clients and denial of service attacks, the Server shall close the oldest unused SecureChannel that has no Session assigned before reaching the maximum number of supported SecureChannels. When Session-less Service invocation is done through a transport mapping that requires the OpenSecureChannel Service, the Server shall maintain a last used time for the SecureChannel to detect the oldest unused SecureChannel."_

### Steps To Reproduce

1. Configure the ReferenceServer.config.xml:
<MaxSessionCount>75</MaxSessionCount>
<MaxChannelCount>76</MaxChannelCount>
2. Setup UACTT project and copying the required certificates to the ReferenceServer's PKI directory.
3. Manual select the tests:
   * Conformance Units -> Security -> Security None
   * Conformance Units -> Security -> Security Aes256-Sha256-RsaPss
4. Run Test and observe the results.

### Environment

```markdown
- OS:Windows 10
- Environment:
- Runtime: .NET8 and .NETFramework 4.8
- Nuget Version: 
- Component:
- Server: ConsoleReferenceServer
- Client: UACTT
```

Siyuan-Xu-Wapice, 2025-02-05:
Hello, @mrsuciu. I noticed the **bug** tag was added then removed from this issue. Is there any update?

Siyuan-Xu-Wapice, 2025-02-05:
Hello, Thank you Suciu for replying. 
I have the latest Windows build of UACTT version 1.04.11-01.00.508. The test environment were both Windows 10 and 11. 
Yes, I configured both UACTT Project and Reference Server/our server (developed by using OPCF'S NET Stack) to have MaxSession=75, MaxChannel = 76.
The nuget version is the latest stable: 1.5.375.443

I noticed that the default ChannelLifetime is set to 300,000 ms, and the UACTT DoS attack 2 test did not wait for half of the channel life timespan. A malicious actor with a trusted ApplicationCertificate could exploit this by continuously creating SecureChannels without establishing a session. The ChannelLifetime/2 cleanup mechanism would be useless in defending the server against such an attack.

mrsuciu, 2025-02-06:
@Siyuan-Xu-Wapice I ran the tests individually for the None profile in debug mode and passed with the attached PR.

Siyuan-Xu-Wapice, 2025-02-12:
@mrsuciu Wonderful! Great efficient work, thanks for the efforts!
Can you help me with a couple of more questions:
1. What's OPCF's normal release cycle of Nuget packages, and when will this PR go to the Release?
2. What's the official term for OPCF's OPCUA development packages, I am using Stack, SDK, Library interchangeably, would be nice to learn the proper name.

PR #2986 (2025-02-11): Remove oldest channel if not used by session, when number of channels reaches MaxChannelCount - 1
## Proposed changes

Close the oldest unused channel prior to consuming the last available slot, thus always allowing one free channel slot to be available.
The pool of potential channels to pick from and chose the oldest is made by channels not used by sessions.

## Related Issues

- Fixes #2979

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixe [... cut]