OPCFoundation/UA-.NETStandard #2983: Opc.Ua.Server SamplingGroup is spawning too many threads.

dardigem, 2025-02-05:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The $SDK\Server\NodeManager\SamplingGroup class from V1.04 or above Opc.Ua.Server library spawns a sperate new thread for each monitoredItem being added to the same sampling group. 

### Expected Behavior

The SamplingGroup is expected to sample all MonitoredItems with same sampling interval on the same thread. It should not spawn separate thread in each call for ApplyChanges.

The $SDK\Server\NodeManager\SamplingGroup class from V1.02 had correct behavior.

Root cause of the issue:
The SamplingGroup::ApplyChanges starts the group even if the group is already running.

Description of fix:
File: $SDK\Server\NodeManager\SamplingGroup 
Ad a new member variable to hold the task that is created in Startup function. Set the default value to null.

Function: Startup
- Store the task returned by Task.Factory.StartNew in the member variable.

Function: ApplyChanges
- Call Startup if (m_items.Count > 0 && null == m_task)

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-02-06:
Good Catch I proposed a fix in #2984

romanett, 2025-02-06:
@dardigem thanks for the hint, not really important as the sampling group is disposed anyway, but good practice

PR #2984 (2025-02-11): [Server] Fix Sampling Group creating a new long running task for every Client AddMonitoredItems Method Call
## Proposed changes

Fix the SamplingGroup of the Server to only initiate startup of the sampling task if it does not exist yet. As in the previous implementation a new sampling task was started on every Client AddMonitoredItems Service Call for a node being handled by the core Node Manager.

## Related Issues

- Fixes #2983 

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancemen [... cut]
closed 2025-02-11 by mrsuciu