OPCFoundation/UA-.NETStandard #2987: Opc.Ua.ServiceResultException: Usage of certificate is not allowed.

charlesaugustineabb, 2025-02-07:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [x] Question
- [x] Help wanted

### Current Behavior

we have a GDS which is having UA Client  trying to connect to OPC Server in a controller device.
OPC Server in Controller is using IDevID (attached below)

Certificate  for the initial  connection . which is created as per  IEC Standard. IDevID is in attachment.
UA Client (in GDS ) while trying to connect with UA Server (in controller ) getting below exception from the SDK code as below.
is it because of key usage content or some other reason? any suggestion , how we can get connected using same  Idev certficate ?

2025-02-05 15:59:34.293 +08:00 [ERR] Certificate Signing Failed for Server PM21-HAORANLI (opc.tcp://PM21-HAORANLI:48013/): Opc.Ua.ServiceResultException: Usage of certificate is not allowed.
 ---> Opc.Ua.ServiceResultException: Usage of certificate is not allowed.
   --- End of inner exception stack trace ---
   at Opc.Ua.CertificateValidator.HandleCertificateValidationException(ServiceResultException se, X509Certificate2 certificate, X509Certificate2Collection chain)
   at Opc.Ua.CertificateValidator.ValidateAsync(X509Certificate2Collection chain, ConfiguredEndpoint endpoint, CancellationToken ct)
   at Opc.Ua.Client.Session.OpenAsync(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain, CancellationToken ct)
   at Opc.Ua.Client.Session.Create(ISessionInstantiator sessionInstantiator, ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, CancellationToken ct)
   at Opc.Ua.Client.DefaultSessionFactory.CreateAsync(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, CancellationToken ct)
   at Opc.Ua.Gds.Client.ServerPushConfigurationClient.Connect(ConfiguredEndpoint endpoint)
   at ABBGlobalDiscoveryServer.PushManagementService.ServerConnect(UserIdentity serveridentity, ServerPushConfigurationClient pushserver) in C:\Users\INPUDC\Downloads\GDSMain\src\ABBGlobalDiscoveryServer\PushManagement\PushManagementService.cs:line 251
   at ABBGlobalDiscoveryServer.PushManagementService.StartServiceToSignApplications() in C:\Users\INPUDC\Downloads\GDSMain\src\ABBGlobalDiscoveryServer\PushManagement\PushManagementService.cs:line 106 {  }

### Expected Behavior

UA CLient in GDS must be able to connect to UA Server which uses IDevID for initial handshaking . after this connection OPC Server  will accept CSR request and GDS client will push signed certificate to Server and it will used  for data communication .
IdevID is used only for initial connection. same connection will used to push certificate using spec12 push interface functions .

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

### Anything else?

[DCA_PKI 1.zip](https://github.com/user-attachments/files/18705342/DCA_PKI.1.zip)
@ @

romanett, 2025-02-07:
Yes the key usage flags are checked, see
https://github.com/OPCFoundation/UA-.NETStandard/blob/ef645f63e773259cdc5eb17c0460f18fbfe74c5c/Stack/Opc.Ua.Core/Security/Certificates/CertificateValidator.cs#L1352-L1358

charlesaugustineabb, 2025-02-08:
Thank u for quick reply.
is there a possibility skip or disable this checking during specific connections? possible to write custom validator?
we would like to use IDevID for initial connection.

romanett, 2025-02-11:
@charlesaugustineabb You could provide an own Certificate Validator. However it seems you just should use your provided certificate as a CA to create IDevID Certificates containing the needed key usage flags as this is an important security feature.
(https://reference.opcfoundation.org/Onboarding/v105/docs/5.1)

EthanChangAED, 2025-02-20:
Converting to Q&A.