OPCFoundation/UA-.NETStandard #2993: How to replace deprecated functin call CheckApplicationInstanceCertificate()

PhilJollans, 2025-02-11:
I have a OPC server implemented with the UI-.NETStandard. I have updated to the most recent version of the NuGet package `OPCFoundation.NetStandard.Opc.Ua` and now I am getting a warning on the statement:
```csharp
await application.CheckApplicationInstanceCertificate(false, 0);
```

> error CS0618: 'ApplicationInstance.CheckApplicationInstanceCertificate(bool, ushort)' is obsolete: 'This method is obsolete since an application now supports different minKey sizes depending on certificate type'

My knowledge about the UI-.NETStandard library is fairly minimal and I can't figure out how I should fix this error. I would be grateful for a some advice, or just a pointer to where I can find suitable documentation.

By the way, this server is only used to simulate another system for the purpose of software tests. In this scenario we are not using any security at all.

Thanks in advance
Phil

romanett, 2025-02-12:
@PhilJollans You just have to remove the second parameter, like this:

`await application.CheckApplicationInstanceCertificate(false);`

PhilJollans, 2025-02-12:
Thank you very much for the fast reply.  
The answer is not quite correct, the correct function call is
```csharp
await application.CheckApplicationInstanceCertificates(false);
```
with a letter 's' on the end of the function name. :smiley:

nietras, 2025-02-12:
This does not seem to be the exact same behavior. When one needs to specify `LifeTimeInMonths` and changes to e.g. (now no min key)
```csharp
    static Task<bool> CheckCertificates(ApplicationInstance applicationInstance) =>
        applicationInstance.CheckApplicationInstanceCertificates(
            silent: false, lifeTimeInMonths: LifeTimeInMonths);
```
then the app, which previously worked fine fails with `'Server does not have an instance certificate assigned.'`,

nietras, 2025-02-12:
I can see the old obsolete method now simply forward to this:
```csharp
        [Obsolete("This method is obsolete since an application now supports different minKey sizes depending on certificate type")]
        public async Task<bool> CheckApplicationInstanceCertificate(
            bool silent,
            ushort minimumKeySize,
            ushort lifeTimeInMonths,
            CancellationToken ct = default)
        {
            return await CheckApplicationInstanceCertificates(silent, lifeTimeInMonths, ct).ConfigureAwait(false);
        }
```
so it appears the new version of the library breaks behavior causing what previusly worked to no longer work, why is that? How can this be fixed? And how does one avoid this breaking usage in the wild?

romanett, 2025-02-12:
@nietras  i will take a look, can you share the configuration File you are using?

nietras, 2025-02-12:
@romanett after some debugging I found out two of the SecurityPolicy's will fail after the update, removing those then allows the server to start, so the relevant section in the config now looks like. Seems like empty security policy uri now no longer is supported or something (silent breaking change).
```xml
        <SecurityPolicies>
            <ServerSecurityPolicy>
                <SecurityMode>SignAndEncrypt_3</SecurityMode>
                <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256</SecurityPolicyUri>
            </ServerSecurityPolicy>
            <ServerSecurityPolicy>
                <SecurityMode>None_1</SecurityMode>
                <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
            </ServerSecurityPolicy>
            <!--<ServerSecurityPolicy>
                <SecurityMode>Sign_2</SecurityMode>
                <SecurityPolicyUri></SecurityPolicyUri>
            </ServerSecurityPolicy>-->
            <!--<ServerSecurityPolicy>
                <SecurityMode>SignAndEncrypt_3</SecurityMode>
                <SecurityPolicyUri></SecurityPolicyUri>
            </ServerSecurityPolicy>-->
        </SecurityPolicies>
```

romanett, 2025-02-12:
@nietras thanks for the insights

nietras, 2025-02-13:
Hmm, perhaps it depends on what certificates are already on the machine, I don't know. I have attached the a full `.Config.xml` that should correspond to our config. Hopefully, that can help.

[OpcUaServer.Config.xml.zip](https://github.com/user-attachments/files/18781295/OpcUaServer.Config.xml.zip)

nietras, 2025-02-13:
Regarding stack trace this simply points to 

https://github.com/OPCFoundation/UA-.NETStandard/blob/cbf789d41e082c10015a4fe5456b70eb9935d4de/Stack/Opc.Ua.Core/Stack/Server/ServerBase.cs#L1384

occurring on:
```csharp
        await applicationInstance.Start(server);
```

zN3utr4l, 2025-02-13:
there is also another thing that no longer goes from version `<PackageReference Include="OPCFoundation.NetStandard.Opc.Ua.Server" Version="1.5.374.158" />` to version `<PackageReference Include="OPCFoundation.NetStandard.Opc.Ua.Server" Version="1.5.375.443" />`.

Error: `Argument 1: Cannot convert from 'Opc.Ua.SecurityConfiguration' to 'Opc.Ua.ApplicationConfiguration'` 
in `ReferenceServer `class
in the line: `certificateValidator.Update(configuration.SecurityConfiguration).Wait();`

```c#
/// <summary>
/// Creates the objects used to validate the user identity tokens supported by the server.
/// </summary>
private void CreateUserIdentityValidators(ApplicationConfiguration configuration)
{
    for (int ii = 0; ii < configuration.ServerConfiguration.UserTokenPolicies.Count; ii++)
    {
        UserTokenPolicy policy = configuration.ServerConfiguration.UserTokenPolicies[ii];

        // create a validator for a certificate token policy.
        if (policy.TokenType == UserTokenType.Certificate)
        {
            // check if user certificate trust lists are specified in configuration.
            if (configuration.SecurityConfiguration.TrustedUserCertificates is not null && configuration.SecurityConfiguration.UserIssuerCertificates is not null)
            {
                CertificateValidator certificateValidator = new();

                certificateValidator.Update(configuration.SecurityConfiguration).Wait();

                certificateValidator.Update(configuration.SecurityConfiguration.UserIssuerCertificates, configuration.SecurityConfiguration.TrustedUserCertificates, configuration.SecurityConfiguration.RejectedCertificateStore);

                // set custom validator for user certificates.
                m_userCertificateValidator = certificateValidator.GetChannelValidator();
            }
        }
    }
}
```

romanett, 2025-02-15:
@zN3utr4l thanks for the hint. We renamed only one of the Methods to UpdateAsync, in the next update both overloads will be called UpdateAsync to be consistent

romanett, 2025-02-15:
> Regarding stack trace this simply points to
> 
> [UA-.NETStandard/Stack/Opc.Ua.Core/Stack/Server/ServerBase.cs](https://github.com/OPCFoundation/UA-.NETStandard/blob/cbf789d41e082c10015a4fe5456b70eb9935d4de/Stack/Opc.Ua.Core/Stack/Server/ServerBase.cs#L1384)
> 
> Line 1384 in [cbf789d](/OPCFoundation/UA-.NETStandard/commit/cbf789d41e082c10015a4fe5456b70eb9935d4de)
> 
>  "Server does not have an instance certificate assigned."); 
> occurring on:
> 
>         await applicationInstance.Start(server);
> that is:
> 
> ```
> This exception was originally thrown at this call stack:
>     Opc.Ua.ServerBase.OnServerStarting(Opc.Ua.ApplicationConfiguration) in ServerBase.cs
>     Opc.Ua.Server.StandardServer.OnServerStarting(Opc.Ua.ApplicationConfiguration) in StandardServer.cs
>     Opc.Ua.ServerBase.Start(Opc.Ua.ApplicationConfiguration) in ServerBase.cs
>     Opc.Ua.Configuration.ApplicationInstance.Start(Opc.Ua.ServerBase) in ApplicationInstance.cs
> ```
> 
> PS: I have no idea how to get a log file, I have tried a lot but cannot get it to output a file e.g. via Config.xml or similar. Including setting a full absolute path, setting 519 as trace mask etc.

@nietras please take a look at the reference Server to see how the logging is set up:

https://github.com/romanett/UA-.NETStandard/tree/master/Applications/ConsoleReferenceServer

The issue occurs if no application certificate is found:

https://github.com/romanett/UA-.NETStandard/blob/23d167abb7fd999b7ffd3c463571e1302aa6e087/Stack/Opc.Ua.Core/Stack/Server/ServerBase.cs#L1378

romanett, 2025-02-17:
@nietras I have no issue running your configuration file with the reference server, maybe an Issue with how you manage Server Startup