OPCFoundation/UA-.NETStandard #3014: Add clilent IP Address or host name to server side session

mheege-abb, 2025-02-24:
### Type of issue

- [ ] Bug
- [x] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

For security reasons, we want to examine the client IP address when a client tries to connect. Please add a property returning the IP address or the host name of the client to Opc.Ua.Server.Session. 

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-02-24:
@mheege-abb I am not shure if this is really the right apporach here, as OPC UA already provides some standardized data model for diagnosing sessions, e.g.  the  [SessionDiagnostics Datatype](https://reference.opcfoundation.org/Core/Part5/v105/docs/12.11) or the  or the [SessionSecurityDiagnostics DataType](https://reference.opcfoundation.org/Core/Part5/v105/docs/12.12).

If you want to injection additional verification in the Session Activation process, I would suggest overriding the ActivateSession Method of the SessionManager. I could also think of adding a "OnBeforeActivate" Callback to the SessionManager class.

mheege-abb, 2025-02-24:
@romanett Thanks for you quick reply.

For normal communication purposes, I would agree with your suggestion. However, we require the IP address for client validation during trust establishment via the OPC UA part 12 push sequence. At the point in time where we want evaluate the client's IP address, we do not yet trust the client's certificate. The information in the Session[Security]Diagnostics data type that might be helpful for us is SecurityDiagnostics.EndpointUrl. However, this information is not secure since is transferred either in plain text or secured with an untrusted certificate.

Therefore we need to know the IP address of the client. This information comes from the socket itself. If this is modified by an attacker, the client will not get responses and the connect attempt will fail.

romanett, 2025-02-24:
@mheege-abb not shure how to wire this up without a tight coupling to the transport channel. I would suggest (if you use Tcp Transport) to override the TcpServerChannel HandleIncomingMessage method & the ProcessOpenSecureChannelRequest to already block clients there? Are you shure none of the OPC UA specified security Mechanisms can help here? Are you familiar with the DeviceOnboarding spec Part 21(https://reference.opcfoundation.org/Onboarding/v105/docs/)(I am not, so not shure if it is helpful in this case).

mregen, 2025-02-25:
I *think* the information you need is e.g. in the TcpMessageSocket. They expose a LocalEndpoint and RemoteEndpoint property. The socket is wired up to the TcpServerChannel. But since the channel implementation is different by transport (tcp, https), it would require a common property to get the information to the server session implementation. 
Is the GDS spec demanding the IP check or is it just an additional  check that you want to add? I am not familiar with latest releases.

mheege-abb, 2025-02-27:
@mregen thanks for the quick reply. The GDS spec does not demand IP check, but since establishing trust is a vulnerable process, investigating client information that is not easily spoofable can be quire important.

romanett, 2025-02-27:
@mheege-abb we will discuss this internally. However significant changes would be needed to achieve this on the library level.

mheege-abb, 2025-02-28:
@romanett and @mregen Thanks for your support. My solution would lead to a straightforward and easy to us interface, but I agree with you that this would not be easy to implement and it would solve a problem that is quite special. I envision a more generic approach but I am currently too deep into other topics to think through it enough to write it down. I am fine if you don't discuss this issue for now. Please keep the bug open for a month. Once I have time, I add a new comment with my proposal. Thanks again for being so responsive.

closed 2025-05-03 by romanett