OPCFoundation/UA-.NETStandard #3017: Null Reference exception after upgrading from 1.5.374.78 to 1.5.374.158

kirstenkgo, 2025-02-24:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [x] Help wanted

### Current Behavior

While testing my changes after the upgrade, I got a null reference exception when trying to open a new OPCUA session. The exception is being thrown from OPC code by the method GetIssuerNoExceptionAsync in CertificateValidator and seems to be related to the server certificate not being from a trusted authority. The exception also occurred when I tried using version 1.5.374.118. 

The only time the exception does not happen is when I connect to the server with no security configured and as an anonymous user (user identity is null). The exception was not occurring before the upgrade.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-02-25:
@kirstenkgo Thank you for reporting this, can you provide a stack trace?

kirstenkgo, 2025-02-25:
System.NullReferenceException: 'Object reference not set to an instance of an object.'

Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.GetIssuerNoExceptionAsync(System.Security.Cryptography.X509Certificates.X509Certificate2 certificate, Opc.Ua.CertificateIdentifierCollection explicitList, Opc.Ua.CertificateStoreIdentifier certificateStore, bool checkRecovationStatus) Line 858	
	Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.GetIssuersNoExceptionsOnGetIssuer(System.Security.Cryptography.X509Certificates.X509Certificate2Collection certificates, System.Collections.Generic.List<Opc.Ua.CertificateIdentifier> issuers, System.Collections.Generic.Dictionary<System.Security.Cryptography.X509Certificates.X509Certificate2, Opc.Ua.ServiceResultException> validationErrors) Line 568	
	Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.InternalValidateAsync(System.Security.Cryptography.X509Certificates.X509Certificate2Collection certificates, Opc.Ua.ConfiguredEndpoint endpoint, System.Threading.CancellationToken ct) Line 929

Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection chain, Opc.Ua.ConfiguredEndpoint endpoint) Line 520	

Opc.Ua.Client.dll!Opc.Ua.Client.Session.Open(string sessionName, uint sessionTimeout, Opc.Ua.IUserIdentity identity, System.Collections.Generic.IList<string> preferredLocales, bool checkDomain) Line 1516	

Opc.Ua.Client.dll!Opc.Ua.Client.Session.Open(string sessionName, uint sessionTimeout, Opc.Ua.IUserIdentity identity, System.Collections.Generic.IList<string> preferredLocales) Line 1499

romanett, 2025-02-27:
@kirstenkgo the line numbers do neither match with 374.118 nor 374.158 please provide some more detail for further anlysis ( configuration, used certs (self signed etc). Which parameter is null? & also a call stack with line numbers matching a released version.

kirstenkgo, 2025-02-27:
@romanett I was testing version 1.5.375.443 to see if the exception happened there also and forgot to change it back. Here is the stack trace for version 374.158

System.NullReferenceException: 'Object reference not set to an instance of an object.'

Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.GetIssuerNoExceptionAsync(System.Security.Cryptography.X509Certificates.X509Certificate2 certificate, Opc.Ua.CertificateIdentifierCollection explicitList, Opc.Ua.CertificateStoreIdentifier certificateStore, bool checkRecovationStatus) Line 1026	
	Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.GetIssuersNoExceptionsOnGetIssuer(System.Security.Cryptography.X509Certificates.X509Certificate2Collection certificates, System.Collections.Generic.List<Opc.Ua.CertificateIdentifier> issuers, System.Collections.Generic.Dictionary<System.Security.Cryptography.X509Certificates.X509Certificate2, Opc.Ua.ServiceResultException> validationErrors) Line 901	
	Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.InternalValidateAsync(System.Security.Cryptography.X509Certificates.X509Certificate2Collection certificates, Opc.Ua.ConfiguredEndpoint endpoint, System.Threading.CancellationToken ct) Line 1143 

Opc.Ua.Core.dll!Opc.Ua.CertificateValidator.Validate(System.Security.Cryptography.X509Certificates.X509Certificate2Collection chain, Opc.Ua.ConfiguredEndpoint endpoint) Line 554	

Opc.Ua.Client.dll!Opc.Ua.Client.Session.Open(string sessionName, uint sessionTimeout, Opc.Ua.IUserIdentity identity, System.Collections.Generic.IList<string> preferredLocales, bool checkDomain) Line 2360	

Opc.Ua.Client.dll!Opc.Ua.Client.Session.Open(string sessionName, uint sessionTimeout, Opc.Ua.IUserIdentity identity, System.Collections.Generic.IList<string> preferredLocales) Line 2329

kirstenkgo, 2025-02-27:
Security Policy: Basic256Sha256

Security Message Mode: Sign

Anonymous user

romanett, 2025-02-28:
@kirstenkgo thanks for the update.

I created a Pull Request to fix the issue.

In the meantime you could configure a valid issuer store in the configuration, then the exception will go away.

PR #3023 (2025-03-28): Enforce Valid Certificate Store configuration
## Proposed changes

If issuer store or rejected store are not configured null reference exceptions could occur.

To avoid these add checks on the Validate method of the Security Configuration to catch invalid configured certificate stores already on startup.

## Related Issues

- Fixes #3017

## Types of changes

- [ ] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [x] Breaking change (fix or feature that wou [... cut]
closed 2025-03-28 by mrsuciu