OPCFoundation/UA-.NETStandard #3018: Subscription FastDataChangeCallBack EventHandler will catch any Exception - also any Exception of further subscribed 'custom code'

KircMax, 2025-02-26:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

[image]
Let's say I have a Subscription and I want to check some things on the Notifications - e.g.:
that there is no non-good statuscode on the items (or even not any bad ones since goodoverload might be acceptable).
Therefore I attach to the EventHandler for the FastDataChangeCallback.
In my custom Method I throw an exception.
[image]
In the Handler itself that Exception will be caught ... 
It has not been straightforward for me that when I use that and throw an Exception it will be caught within the Subscription -> OnMessageReceivedAsync:

[image]

https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Libraries/Opc.Ua.Client/Subscription/Subscription.cs#L2399

### Expected Behavior

Maybe only some sort of Exceptions should be caught within the OnMessageReceivedAsync ? 

### Steps To Reproduce

1. Create a subscription
2. Attach to the FastDataChangeCallback
3. throw an exception within the attached method

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

KircMax, 2025-02-26:
Probably it is also bad style that I do not just set a flag within my handler about the nongood status codes, etc.
and I can just do that right now but I wanted to raise awareness that this has not felt completely straightforward for me. 
Prob. a skill issue on my side when working with EventHandler, etc.

romanett, 2025-02-27:
To me this catch all seems to be good practice to handle e.g. bad event handlers throwing exceptions without allowing them to block the processing of further messages.

KircMax, 2025-02-27:
-> as intended

closed 2025-02-27 by KircMax