OPCFoundation/UA-.NETStandard #3027: Endpoint Discovery fails calling SelectEndpoint

benjsherman, 2025-03-04:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [x] Help wanted

### Current Behavior

In [UAClient.cs](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/UAClient.cs) on line 200, this fails when trying to connect to the same OPCUA server in succession. 
Here is the stack trace:
``` cs
Opc.Ua.ServiceResultException: Error establishing a connection: BadNotConnected
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request)
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints)
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris)
   at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(ApplicationConfiguration application, String discoveryUrl, Boolean useSecurity, Int32 discoverTimeout)
   at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(ApplicationConfiguration application, String discoveryUrl, Boolean useSecurity)
```

### Expected Behavior

I expect to be able create as connections as my OPCUA server will allow without any errors. I don't expect it to error during endpoint discovery.

### Steps To Reproduce

Replicate by setting up two unit tests each connecting to the same OPCUA server, the first conneciton will succeed, the second connection will fail. If run separately both will succeed.

Also interesting is adding a `Thread.Sleep` of one second in the Dispose method prevents this error

### Environment

```markdown
- OS: Windows 11
- Environment:
- Runtime:x64
- Nuget Version: 1.5.375.457
- Component:
- Server:
- Client: x
```

romanett, 2025-03-05:
@benjsherman Thank you for reporting a potential bug. Please provide the exact sequence of function calls to the UAClient.cs or even better a test project.

benjsherman, 2025-03-05:
I've attached a project with unit tests for replication of my issue:

[Code.zip](https://github.com/user-attachments/files/19095869/Code.zip)

Solution file is under Code/OPCUACLient/OPCUAClient.sln

Failing function call is documented in OpcDataService.cs Line 129

OpcFoundationTest.cs has two test that when ran in batch, one will fail. 
There are two lines to sleep after doing the OPCUA read that when uncommented, permits both tests to pass when run in batch.

You'll need a valid OPCUA server to connect to and this test scenario was against an un-signed, un-secured, no credential OPCUA server.

romanett, 2025-03-16:
@benjsherman I cannot reproduce this with our reference server, could be a server issue. Please reopen if you have a reproduction scenario with our reference server.

closed 2025-03-16 by romanett

perclausen, 2025-03-16:
> [@benjsherman](https://github.com/benjsherman) I cannot reproduce this with our reference server, could be a server issue. Please reopen if you have a reproduction scenario with our reference server.

Think its the same issue as #2671 ?

benjsherman, 2025-03-24:
I wasn't able to replicate this issue with the reference server. This must be something with the Ignition OPC UA implementation. Work around for anyone who stumbles across this is to cache endpoint discovery data so that consecutive back to back connections don't have to run into this scenario and `CoreClientUtils.SelectEndpoint(application.ApplicationConfiguration, discoveryUri, useSecurity);`  is only called once.