OPCFoundation/UA-.NETStandard #3028: Null Reference Exception in UaSCUaBinaryChannel.WriteSymmetricMessage after Client stop

romanett, 2025-03-05:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The UaSCUaBinaryChannel.WriteSymmetricMessage throws an exception here: https://github.com/OPCFoundation/UA-.NETStandard/blob/ef9387b525b39fce08d5e83377b68de691d99489/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryChannel.Symmetric.cs#L613
  as the token is null.

This occurs when the Client is shut down.

### Expected Behavior

No Exception is thrown.

Maybe 0? should be written if no token is present?

### Steps To Reproduce

1. Start Reference Server
2. Start Reference Client
3. Stop Reference Client while communication is in progress
4. Exception is thrown
https://github.com/OPCFoundation/UA-.NETStandard/blob/ef9387b525b39fce08d5e83377b68de691d99489/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryChannel.Symmetric.cs#L613

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: Net 9
- Nuget Version: Current Master
- Component: Server
- Server: Current Master Reference Server
- Client: Current Master Reference Client
```

### Anything else?

MesageBody: {Opc.Ua.ServiceFault}
MessageType: 4674381
isRequest: false

romanett, 2025-03-05:
Could be related to Token Dispose PR

romanett, 2025-03-12:
This TCP message type is none of the known OPC UA Message Types

romanett, 2025-03-12:
This seems to be a general issue in handling this kind of disconnect.

If the line above is fixed the symmetric sign fails:

[image]

IMHO the server shall just stop processing the message, as it is not needed any more

mregen, 2025-03-17:
Could be a race condition on close, do you have a call stack?

romanett, 2025-03-17:
[image]