OPCFoundation/UA-.NETStandard #3029: 如何使用C#代码创建结构体，并创建结构体的引用变量

1924006212, 2025-03-06:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [x] Question
- [ ] Help wanted

### Current Behavior

```
protected override NodeStateCollection LoadPredefinedNodes(ISystemContext context)
{
    NodeStateCollection nodeStateCollection = new NodeStateCollection();
    //FolderState root = CreateFolder(null, null, "Root");
    //root.AddReference(ReferenceTypes.Organizes, true, ObjectIds.ObjectsFolder); // 将节点添加到服务器根节点
    //root.EventNotifier = EventNotifiers.SubscribeToEvents;
    ushort namespaceIndex = base.NamespaceIndex;
    FolderState root2 = CreateFolder(null, null, "Root2");
    root2.AddReference(ReferenceTypes.Organizes, true, ObjectIds.ObjectsFolder); // 将节点添加到服务器根节点
    root2.EventNotifier = EventNotifiers.SubscribeToEvents;
    
    
    BaseObjectState PLC = new BaseObjectState(root2);
    PLC.SymbolicName = "PLC_B";
    PLC.TypeDefinitionId = ObjectTypeIds.BaseObjectType;
    PLC.NodeId = new NodeId("ns=" + namespaceIndex + $";s={PLC.SymbolicName}");
    PLC.BrowseName = new QualifiedName("CustomObject", namespaceIndex);
    PLC.DisplayName = new LocalizedText("en-US", "PLC_B");
    PLC.EventNotifier = EventNotifiers.None;
    PLC.AddReference(ReferenceTypes.Organizes, true, ObjectIds.ObjectsFolder);
    PLC.EventNotifier = EventNotifiers.SubscribeToEvents;

    //BaseDataVariableState PLCStruct = new BaseDataVariableState(root2);
    //PLCStruct.SymbolicName = "PC-PLCStruct";
    //PLCStruct.TypeDefinitionId = DataTypeIds.vari;
    //PLCStruct.NodeId = new NodeId("ns=" + namespaceIndex + $";s={PLCStruct.SymbolicName}");
    //PLCStruct.BrowseName = new QualifiedName("CustomObject", namespaceIndex);
    //PLCStruct.DisplayName = new LocalizedText("en-US", "PC-PLCStruct");
    //PLCStruct.AddReference(ReferenceTypes.Organizes, true, ObjectIds.OPCBinarySchema_TypeSystem);
    //nodeStateCollection.Add(PLCStruct);

    BaseDataVariableState Struct = new BaseDataVariableState(root2);
    Struct.SymbolicName = "PC-PLCStruct";
    Struct.TypeDefinitionId = VariableTypes.DataTypeDescriptionType;
    Struct.NodeId = new NodeId("ns=" + namespaceIndex + $";s={Struct.SymbolicName}");
    Struct.BrowseName = new QualifiedName("CustomObject", namespaceIndex);
    Struct.DisplayName = new LocalizedText("en-US", "PC-PLCStruct");
    Struct.DataType=DataTypeIds.String;
    Struct.ValueRank= ValueRanks.Scalar;
    Struct.Value = "CustomSensorData";
    Struct.AddReference(ReferenceTypes.HasDescription, true, ObjectIds.DefaultBinary);
    Struct.AddReference(ReferenceTypes.Organizes, true, ObjectIds.OPCBinarySchema_TypeSystem);
    nodeStateCollection.Add(Struct);

    BaseObjectState customObject = new BaseObjectState(root2);
    customObject.SymbolicName = "CustomObject";
    customObject.TypeDefinitionId = DataTypes.BaseDataType;
    customObject.NodeId = new NodeId("ns=" + namespaceIndex + ";s=CustomObject");
    customObject.BrowseName = new QualifiedName("CustomObject", namespaceIndex);
    customObject.DisplayName = new LocalizedText("en-US", "Custom Object");
    customObject.EventNotifier = EventNotifiers.None;
    AddRootNotifier(customObject);

    BaseDataVariableState<Person> customVariable = new BaseDataVariableState<Person>(null);
    customVariable.SymbolicName = "CustomVariable";
    //customVariable.TypeDefinitionId = VariableTypeIds.BaseDataVariableType;
    customVariable.NodeId = new NodeId("ns=" + namespaceIndex + ";s=CustomVariablePerson");
    customVariable.BrowseName = new QualifiedName("CustomVariable", namespaceIndex);
    customVariable.DisplayName = new LocalizedText("en-US", "Custom VariablePerson");
[... 202 lines cut ...]

[image]

[image]

但是现在是这样的

[image]

[image]

[image]
现在我创建的结构体类型的变量但是，不显示结构体的值

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-03-06:
@1924006212 please provide your description in english

EthanChangAED, 2025-03-06:
Converting to Q&A for visibility.