OPCFoundation/UA-.NETStandard #3030: Password available in plain string format in virtual memory

Nikhil-in1abb, 2025-03-06:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [x] Question
- [ ] Help wanted

### Current Behavior

I am using a client application to connect with OPC UA server that supports username and password login.

In my client when i receive the input of the username and password I store it in an encrypted format.

when i try to connect to the server I use the below code 

userIdentity = new UserIdentity(usernameStr, decryptedPasswordStr);       

   var session = Opc.Ua.Client.Session.Create(
                    configuration,
                    endpoint,
                    false,
                    false,
                    sessionName,
                    sessionTimeout,
                    userIdentity,
                    preferredLocales,
                    default(CancellationToken)).Result;

As it is shown I am making a local copy  after decrypt in a string variable decryptedPasswordStr

This string is stored in memory in plaintext and it is in read only format. It will not be cleared. If I use tools like process hacker, I will be able to read the password in plain string text making it exposed and vulnerable.

### Expected Behavior

Are there overloads to send password in encrypted format/byte format. if so I would request example code snippet to achieve the same.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

marcschier, 2025-03-06:
@Nikhil-in1abb , you should be able to implement IUserIdentity to implement it in the way you want.   Does that work?

Nikhil-in1abb, 2025-03-07:
The IUseridentity class does not provide sufficient members to achieve my intended result.

I would want to use encoded/encrypted password to create new UserIdentity.

There is one other approach I tried. as shown below:

                        UserNameIdentityToken token = new UserNameIdentityToken();
                        token.UserName = usernameStr;
                        token.Password = Encoding.UTF8.GetBytes(DecryptedPasswordStr);
                        
                        userIdentity = new UserIdentity(token);

when I try to create session with this I am getting 'BadIdentityTokenInvalid' returned from the OPC UA server.

I would need an example/support to pass the token with password and use encoding/encryption instead of using 
token.DecryptedPassword

romanett, 2025-03-10:
@Nikhil-in1abb Microsoft recommends not to use SecureString any more. (https://github.com/dotnet/platform-compat/blob/master/docs/DE0001.md).
Instead for environments with high security needs you could use Certificate or Token User Identities.

For securing the memory of the running process maybe virtualization-based security (VBS) could be an option.

What Method of storing /access passwords in the user identity token would you suggest?

Nikhil-in1abb, 2025-03-10:
@romanett, In my case my target OPC UA server is a legacy controller that supports only username/password user token policy. 
so creating secure tokens like certificate based authentication/issued token is not an option.

There is an overload to use token to create UserIdentityToken (code section pasted above) where in I can use token.password which is byte array. Unlike string which is immutable in c# we could use byte array that can be overwritten/cleared upon execution.

when I try to execute create session with this I am getting 'BadIdentityTokenInvalid' returned from the OPC UA server.

I would need an example/support to pass the token with password instead of using token.DecryptedPassword

romanett, 2025-03-11:
@Nikhil-in1abb thanks for the update.

The relevant code seems to be here:

https://github.com/OPCFoundation/UA-.NETStandard/blob/fe0aec4969729383b33b1cf58e38dffb483cdb81/Libraries/Opc.Ua.Server/Session/Session.cs#L877

The Session implementation in the server is tightly bound to UserIdentityToken. You need to add a custom override for UserIdentityToken that stores the PW in a manner you need. And implemements encrypt / decrypt methods of UserIdentityToken. 

As a short term solution I would suggest overriding the existing UserNameIdentityToken:

https://github.com/OPCFoundation/UA-.NETStandard/blob/fe0aec4969729383b33b1cf58e38dffb483cdb81/Stack/Opc.Ua.Core/Stack/Types/UserIdentityToken.cs#L91

As a longer term solution I would like to extend the existing UserNameIdentityToken in the stack with extra property / constructors to accept PW as byte array.

romanett, 2025-03-12:
@Nikhil-in1abb I finally had time to debug it myself. Seems the issue is, that the password is not even sent from the client to the server if it is stored as byte array

Nikhil-in1abb, 2025-03-12:
@romanett , I had tried a work around by implementing the below override, this will copy the password bytes instead of decrypted password string. But still i see some password available in virtual memory I suspect either  Utils.Append, or SecurityPolicies.Encrypt to make a internal copy of the byte array that is not handled properly.

Correct me if I am wrong, I would need guidance to solve the issue

public partial class ExtendedIdentityToken : UserNameIdentityToken
{
    public ExtendedIdentityToken(string username, byte[] password)
        : base()
    {
        UserName = username;
        Password = password;
        EncryptionAlgorithm = null;
    }

    public override void Encrypt(X509Certificate2 certificate, byte[] senderNonce, string securityPolicyUri)
    {
        if (Password == null || Password.Length == 0)
        {
            Password = null;
            return;
        }

        if (string.IsNullOrEmpty(securityPolicyUri) || securityPolicyUri == SecurityPolicies.None)
        {
            return; 
        }
 
        //byte[] dataToEncrypt = Utils.Append(Password, senderNonce);

        EncryptedData encryptedData = SecurityPolicies.Encrypt(
            certificate,
            securityPolicyUri,
            dataToEncrypt);

        Password = encryptedData.Data;
        EncryptionAlgorithm = encryptedData.Algorithm;

    }

closed 2025-04-03 by mrsuciu