OPCFoundation/UA-.NETStandard #3032: Wrong error thrown "BadTooManyArguments" instead of "BadNotImplemented"

raserle, 2025-03-06:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [x] Question
- [ ] Help wanted

### Current Behavior

@romanett 
In [Bug](https://github.com/OPCFoundation/UA-.NETStandard/issues/3024) you see the error "BadTooManyArguments" on call the method. 
If you have a "NodeManager:CustomNodeManager2" with Call method
[image]

and call the base 
[image]

you end in MethodState
[image]
and on line 653 is the return. That is what happen wenn no method implemented. 

The correct check what we need, comes in this call.
[image]

That is a test for implementation of method, and this could return NotImplemented. But with the checks before we never reach this.
[image]

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-03-06:
@raserle Thanks for bringing this up
MethodState is initialized with (User)Exectuable true. 
And I don`t think we can change it without breaking existing implementations.
So not shure if this StatusCode would ever be thrown, even if we put the check before the input arguments validation.

raserle, 2025-03-06:
Thats the body of the second Call Method, that get called after check input/output params 
```
GenericMethodCalledEventHandler2 onCallMethod2 = OnCallMethod2;
if (onCallMethod2 != null)
{
    return onCallMethod2(context, this, objectId, inputArguments, outputArguments);
}
GenericMethodCalledEventHandler onCallMethod = this.OnCallMethod;
if (onCallMethod != null)
{
    return onCallMethod(context, this, inputArguments, outputArguments);
}
if (Executable && UserExecutable)
{
    return StatusCodes.BadNotImplemented;
}
return StatusCodes.BadUserAccessDenied;
```

Is it possible to test, this before run argument tests?
```
GenericMethodCalledEventHandler2 onCallMethod2 = OnCallMethod2;
GenericMethodCalledEventHandler onCallMethod = this.OnCallMethod;

if (onCallMethod2 == null && onCallMethod == null) return NotImplemented....
```

romanett, 2025-03-16:
@raserle This fix is not really gonna work as most overrides add an own more specific call handler instead of using the default one. And most methods dont override this Call method. This would break existing implementations. Having not implemented methods in your nodeset avialable for clients to call could be considered bad practice.

closed 2025-03-16 by romanett