OPCFoundation/UA-.NETStandard #3041: DataItemType and its sub types have incorrect ValueRank

saurla, 2025-03-27:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

DataItemType has ValueRank 0 = OneOreMoreDimensions. Also all sub types of DataItemType have ValueRank 0. See attached image illustrating the problem [image].

### Expected Behavior

Based on the specification (https://reference.opcfoundation.org/Core/Part8/v105/docs/5.3.1), DataItemType should have ValueRank -2 = Any. 

### Steps To Reproduce

1. Compile ConsoleReferenceServer
2. Start ConsoleReferenceServer
3. Connect to ConsoleReferenceServer
4. Read ValueRank Attribute of the DataItemType node (i=2365)
5. Observe invalid ValueRank 0

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .NET
- Nuget Version: .NET 9.0
- Component: Not sure where the problem is, might be Opc.Ua.Core or UA-ModelCompiler
- Server: ConsoleReferenceServer
- Client: Ua Expert
```

saurla, 2025-03-27:
Based on quick investigation I think it's the same problem that was already once fixed: https://github.com/OPCFoundation/UA-.NETStandard/issues/750.

In the UA-ModelCompiler there was a fix in this commit https://github.com/OPCFoundation/UA-ModelCompiler/commit/5f2afa3f388efee5596ffa874b046a18ade42246#diff-8895de6ce47cd3e1f1cfbb067696a6e97e5fda1ec30accf02f5228defe66a230R1062 which fixed `GetAttributesToSave `methods in `BaseVariableState `and `BaseVariableTypeState` to persist ValueRank correctly to initialization string. At that point UA-ModelCompiler used its own copy of the Opc.Ua.Core. Now it seems to be linked to this repository as sub-module and the fix most likely was never done to this repo.

So most likely to fix this following is needed:

- Make the same fix to this repository
- Update UA-ModelCompiler to use the version of this repository where fix is done
- Generate node sets
- Take node sets in to use in Opc.Ua.Core

saurla, 2025-03-27:
More investigation shows that ValueRank is not written to InitializationString if its value is Any. This is in a way fine since it also won't be read from the InitializationString and instead value returned by the `GetDefaultValueRank()` of the type is used. So instantiating DataItemState in code works fine. 

Problem is with the UA-ModelCompiler when it compiles the nodeset file Opc.Ua.PredefinedNodes.uanodes. The ValueRank there is left to default int value 0 because `GetAttributesToSave()` does not return ValueRank attribute when ValueRank is -2. 

I tested the solution I suggested in previous comment and was able to generate Opc.Ua.PredefinedNodes.uanodes when the ValueRank was correctly -2.

saurla, 2025-03-27:
I have written a test case to reproduce the issue:

```using System;
using System.Collections.Generic;
using System.IO;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using Iced.Intel;
using NUnit.Framework;
using static Opc.Ua.NodeState;
using Assert = NUnit.Framework.Legacy.ClassicAssert;

namespace Opc.Ua.Core.Tests.Stack.State
{
    /// <summary>
    /// Tests serialization and deserialization of BaseDataVariableTypeState
    /// </summary>
    [TestFixture]
    [SetCulture("en-us"), SetUICulture("en-us")]
    public class BaseDataVariableTypeStateSerializationTest
    {
        [Test]
        public void ValueRankPersistingTest(
            [Values(ValueRanks.ScalarOrOneDimension, ValueRanks.Any, ValueRanks.Scalar,
            ValueRanks.OneOrMoreDimensions, ValueRanks.OneDimension, ValueRanks.TwoDimensions)] int valueRank)
        {
            var variable = new BaseDataVariableTypeState();
            var serviceMessageContext = new ServiceMessageContext();

            var systemContext = new SystemContext() { NamespaceUris = serviceMessageContext.NamespaceUris };

            variable.Create(
                    new SystemContext() { NamespaceUris = serviceMessageContext.NamespaceUris },
                    VariableTypeIds.DataItemType,
                    BrowseNames.DataItemType,
                    new LocalizedText("DataItemType"),
                    true);

            variable.ValueRank = valueRank;

            var loadedVariable = new BaseDataVariableTypeState();

            using (MemoryStream stream = new MemoryStream())
            {
                variable.SaveAsBinary(systemContext, stream);
                
                stream.Position = 0;
                loadedVariable.LoadAsBinary(systemContext, stream);
            }

            Assert.AreEqual(valueRank, loadedVariable.ValueRank);
        }
    }
}
```

saurla, 2025-03-27:
I also found out another potential way to fix this which does not require regeneration of nodeset files. It could be possible to add `Initialize(context)` call here https://github.com/OPCFoundation/UA-.NETStandard/blob/2f0770c569846d2f51849f6045d2f7091eb03898/Stack/Opc.Ua.Core/Stack/State/NodeState.cs#L681, which would make sure all values are set to default values read from the InitializationString before reading values from the BinaryDecoder. However, I'm not sure if there would be any unwanted side effects from the `Initialize `call.

saurla, 2025-03-27:
If needed I'm ready to work on this issue and submit a PR when there is a decision how to fix the issue.

saurla, 2025-03-27:
The `Initialize(context)` call inside `LoadAsBinary` was not good idea after all since Initialize of generated classes will also call `LoadAsBinary` causing an infinite recursion.

mrsuciu, 2025-03-28:
@saurla Thank you for raising the issue and for the investigation. Stay tuned until a decision on which the best approach is decided.

PR #3049 (2025-04-10): Set default value for the ValueRank attribute of BaseVariableState and BaseVariableTypeState to ValueRanks.Any as in the specification
Set default value for the ValueRank attribute of BaseVariableState and BaseVariableTypeState types and subtypes to ValueRanks.Any as specified in the specification.

## Proposed changes

The default value used for the ValueRank attribute for BaseVariableState and BaseVariableTypeState type and subtype instances should be ValueRanks.Any. The attribute is omitted during serialization for efficiency purpose when the value is the default ValueRanks.Any,  but must b [... cut]
closed 2025-04-10 by mrsuciu