OPCFoundation/UA-.NETStandard #3045: XML Encoding of Matrix is Incorrect

kevinherron, 2025-04-06:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The way `XmlEncoder` encodes Matrix values is incorrect.

In Part 6 the schema from a Matrix represented by a Variant is defined as:
```
<tns:Matrix>
  <tns:Dimensions>
    <tns:Int32>2</tns:Int32>
    <tns:Int32>2</tns:Int32>
   </tns:Dimensions>
  <tns:Elements>
    <tns:String>A</tns:String>
    <tns:String>B</tns:String>
    <tns:String>C</tns:String>
    <tns:String>D</tns:String>
    </tns:Elements>
</tns:Matrix>
```

Which means the expected output for something like:
```
var context = new ServiceMessageContext();
string actualXmlValue;

using (IEncoder xmlEncoder = new XmlEncoder(context))
{
    var value = new Matrix(
        new int[] { 1, 2, 3, 4 },
        BuiltInType.Int32,
        new int[] { 2, 2 }
    );

    xmlEncoder.PushNamespace(Namespaces.OpcUaXsd);
    xmlEncoder.WriteVariant("Test", new Variant(value));
    xmlEncoder.PopNamespace();
    actualXmlValue = xmlEncoder.CloseAndReturnText();
}
```

would be:
```
<Test xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <Value>
    <Matrix>
      <Dimensions>
        <Int32>2</Int32>
        <Int32>2</Int32>
      </Dimensions>
      <Elements>
        <Int32>1</Int32>
        <Int32>2</Int32>
        <Int32>3</Int32>
        <Int32>4</Int32>
      </Elements>
    </Matrix>
  </Value>
</Test>
```

yet this implementation of `XmlEncoder` yields this instead:
```
<Test xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <Value>
    <Matrix>
      <Elements>
        <ListOfInt32>
          <Int32>1</Int32>
          <Int32>2</Int32>
          <Int32>3</Int32>
          <Int32>4</Int32>
        </ListOfInt32>
      </Elements>
      <Dimensions>
        <Int32>2</Int32>
        <Int32>2</Int32>
      </Dimensions>
    </Matrix>
  </Value>
</Test>
```

Note:
- there is an extra "ListOfInt32" wrapper element
- the "Dimensions" element appears after "Elements" instead of before

### Expected Behavior

Expect:
```
<Test xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <Value>
    <Matrix>
      <Dimensions>
        <Int32>2</Int32>
        <Int32>2</Int32>
      </Dimensions>
      <Elements>
        <Int32>1</Int32>
        <Int32>2</Int32>
        <Int32>3</Int32>
        <Int32>4</Int32>
      </Elements>
    </Matrix>
  </Value>
</Test>
```

Actual:
```
<Test xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <Value>
    <Matrix>
      <Elements>
        <ListOfInt32>
          <Int32>1</Int32>
          <Int32>2</Int32>
          <Int32>3</Int32>
          <Int32>4</Int32>
        </ListOfInt32>
      </Elements>
      <Dimensions>
        <Int32>2</Int32>
        <Int32>2</Int32>
      </Dimensions>
    </Matrix>
  </Value>
</Test>
```

### Environment

```markdown
Latest? git checkout, 66ba504751990520f070f1359788c02ef225c14c
```

romanett, 2025-04-07:
The implementation can be found here: https://github.com/OPCFoundation/UA-.NETStandard/blob/66ba504751990520f070f1359788c02ef225c14c/Stack/Opc.Ua.Core/Types/Encoders/XmlEncoder.cs#L2168

romanett, 2025-04-14:
This also seems to affect the decoder:

https://github.com/OPCFoundation/UA-.NETStandard/blob/dbb9acba6cc99943c4ab80cf7d3b9df2b4b4368c/Stack/Opc.Ua.Core/Types/Encoders/XmlDecoder.cs#L2729

romanett, 2025-04-14:
@kevinherron The IEncoder Methods WriteArray/ReadArray serialize the matrix according to spec. This could be a workaround in the meantime. I will investigate if the other method is kept this way for iop reasons

kevinherron, 2025-04-14:
I am not in need of workaround, but I think the issue with the proposed workaround is that `WriteMatrix` method is used while encoding a `Variant`, which is likely to be encoded as part of something else, so it seems unlikely the caller/user would even have an opportunity to do something different.

romanett, 2025-04-15:
@kevinherron you are right, I proposed a fix in the linked PR

PR #3053 (2025-04-15): Fix XML Encoding and decoding of Matrix Element in Variant to conform to spec
## Proposed changes

A Variant containing a multi dimensional array needs to be encoded with a specified syntax.
The XML Encoder / Decoder of the stack did only follow this Spec for the ReadArray / WriteArray implementation but not ReadVariant/WriteVariant

https://reference.opcfoundation.org/Core/Part6/v105/docs/5.3.1.17

This PR fixes JsonEncoder & Decoder & introduces a test case.

#3045

## Related Issues

- Fixes #3045

## Types of changes

- [ ] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancemen [... cut]
closed 2025-04-15 by mrsuciu