OPCFoundation/UA-.NETStandard #3046: NullReferenceException in Opc.Ua.Server.MasterNodeManager.ValidateRolePermissions()

GhislainPr, 2025-04-07:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

NullReferenceException in Opc.Ua.Server.MasterNodeManager.ValidateRolePermissions() method if Opc.Ua.NodeState.ClearChangeMasks() is called while Opc.Ua.Server.CustomNodeManager2.CreateMonitoredItems() is processing.
In this case, the Session property of MonitoredItem is null.

### Expected Behavior

No exception attempted while changing nodes during monitoreditem creation.

### Steps To Reproduce

It's difficult to reproduce because it's depend on thread synchronization.

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .NET 4.6.2,
- Nuget Version: 1.05.375.457
- Component: Opc.Ua.Server
- Server: Reference Server
- Client: UA Expert
```

### Anything else?

Suggested solution in Libraries/Opc.Ua.Server/Diagnostics/MonitoredNode.cs :
295 : if (monitoredItem.Session != null && monitoredItem.AttributeId == Attributes.Value && (changes & NodeStateChangeMasks.Value) != 0)

instead of :
295 : if (monitoredItem.AttributeId == Attributes.Value && (changes & NodeStateChangeMasks.Value) != 0)

romanett, 2025-04-07:
@GhislainPr I dont see MasterNodeManager.ValidateRolePermissions access the Session, validation should take place with the saved user identity:

https://github.com/OPCFoundation/UA-.NETStandard/blob/66ba504751990520f070f1359788c02ef225c14c/Libraries/Opc.Ua.Server/NodeManager/MasterNodeManager.cs#L3434

MonitoredItem is expected to have a null Session in some cases

GhislainPr, 2025-04-08:
Thanks for your interest.

I try to detail the technical context :
In one thread, _Opc.Ua.NodeState.ClearChangeMasks()_ is called.
In the call stack of the _Opc.Ua.Server.MasterNodeManager.ValidateRolePermissions()_ method is also called (MonitoredNode.cs line 298 : `ServiceResult validationResult = NodeManager.ValidateRolePermissions(new OperationContext(monitoredItem), node.NodeId, PermissionType.Read);`). At this step, A _OperationContext_ is created from a _monitoredItem_, whith this _Session_ property value equals to null. So, the value of the _Identity_ property of _OperationContext_ stills null. The processing continue until the crash in _ValidateRolePermissions()_ method  (MasterNodeManager.cs line 3323 : `var currentRoleIds = context.UserIdentity.GrantedRoleIds;`).

During this time, in another thread, _Opc.Ua.Server.Subscription.CreateMonitoredItems()_ is **locked** (Subscription.cs line 1473 : `lock (m_lock)`). At this step,  MonitoredItems are created, but their **_SubscriptionCallbacbk_ property still at null value**. Their value is set some instructions after in Subscription.cs line 1499 : `monitoredItem.SubscriptionCallback = this;`.

PR #3047 (2025-05-16): [Server] Enhance role permission validation
- Make `ValidateRolePermissions` virtual in `CustomNodeManager.cs` to allow custom validation
- Initialize OperationContext in Node.ReadAttribute of `MonitoredNode` to allow custom validation
- Update `ValidateRolePermissions` in `MasterNodeManager.cs` to prevent null reference exception

## Related Issues

- Fixes #3046

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking chan [... cut]