OPCFoundation/UA-.NETStandard #3063: <title>

paulhfu, 2025-04-23:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [x] Compliance
- [x] Question
- [ ] Help wanted

### Current Behavior

Hello, I hope you can help me with my issue. 
A browse call with a view description containing the default view, given by the None-NodeId returns with "BadNodeNotInView". This only happens, if the timestamp in the view description is unequal to the Windows epoch.

### Expected Behavior

A browse call, containing the default view which is given by the None-NodeId should return success on existing nodes regardless of the settings of the view-version or -timestamp.

### Environment

```markdown
OS: Windows 10
Environment: Visual Studio 2022
Component: Opc.Ua.Core
Server: Reference Server
Client: async-ua
```

### Anything else?

[Here](https://github.com/OPCFoundation/UA-.NETStandard/blob/d5fd96b7c1ff5c19f116847a79c1d00d6e348b27/Stack/Opc.Ua.Core/Stack/Types/ViewDescription.cs#L36) all the view-description members are testet to determine a default view. According to [here](https://reference.opcfoundation.org/Core/Part4/v104/docs/7.39#_Ref141201863), only the None-ness of the NodeId should be tested.

Am I reading the reference manual correclty? Is there another reason or definition I am not aware of?

Any help is appreachiated.
Thank you.

mrsuciu, 2025-04-25:
The same reference states: 

> The current view is used if timestamp is null and viewVersion is 0

https://reference.opcfoundation.org/Core/Part4/v105/docs/7.45#:~:text=The%20current%20view%20is%20used%20if%20timestamp%20is%20null%20and%20viewVersion%20is%200.

And the "empty" ViewDescription field is a bit ambiguous so that is why both 
https://github.com/OPCFoundation/UA-.NETStandard/blob/d5fd96b7c1ff5c19f116847a79c1d00d6e348b27/Stack/Opc.Ua.Core/Stack/Types/ViewDescription.cs#L36

and 

https://github.com/OPCFoundation/UA-.NETStandard/blob/d5fd96b7c1ff5c19f116847a79c1d00d6e348b27/Stack/Opc.Ua.Core/Stack/Types/ViewDescription.cs#L31

Are considered "empty":

> Description of the View to browse (see 7.45 for ViewDescription definition). An empty ViewDescription value indicates the entire AddressSpace. Use of the empty ViewDescription value causes all References of the nodesToBrowse to be returned. Use of any other View causes only the References of the nodesToBrowse that are defined for that View to be returned.

https://reference.opcfoundation.org/Core/Part4/v105/docs/5.9.2.2#:~:text=Description%20of%20the%20View,View%20to%20be%20returned.

opcfoundation-org, 2025-04-25:
In Part 4, no embedded structure can be null so the only representation of an "empty" ViewDescription is all fields set to default values. The possibility of a null only shows up in C# code because the generated code uses class instead of struct for efficiency.

paulhfu, 2025-04-25:
Thank you a lot for this clarification. I was asking, because our testing client threw an error after we replaced our Server SDK with this one. For our test client, we use python with asyncua, where I found [this](https://github.com/FreeOpcUa/opcua-asyncio/issues/654) issue.
So our problem is the client and not the server.

closed 2025-04-25 by paulhfu