OPCFoundation/UA-.NETStandard #3074: BigInteger is not supported in conversions

Ignatella, 2025-05-12:
### Type of issue

- [x] Bug
- [x] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Hello

I noticed that BigInteger (present in .net standard) is not supported in conversions. 

[image]

[image]

Would You welcome a PR adding this ?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2025-05-13:
An implicit conversion rom a BigInteger to float may lose precision. A proper conversion would be to the OPC UA Decimal type or the caller should make sure the conversion to float is properly done. (and being aware of the precision loss)

Ignatella, 2025-05-19:
@mregen Thank You for the response. 

At the same double -> float conversion is done siliently, what make me create this issue. To be consistent it would be great to add BigInteger.

https://github.com/OPCFoundation/UA-.NETStandard/blob/d1f90661ceeb01d22811924978de30b724a3da6e/Stack/Opc.Ua.Core/Types/Utils/TypeInfo.cs#L2148