OPCFoundation/UA-.NETStandard #3126: Disconnecting and reconnecting does not work. After the server is functioning properly, the client cannot reconnect properly

shunnet, 2025-07-25:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Disconnecting and reconnecting does not work. After the server is functioning properly, the client cannot reconnect properly

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

marcschier, 2025-07-25:
@shunnet - Can you provide a repro, step by step, please?

shunnet, 2025-07-25:
`clientSession.KeepAliveInterval = 2000; 
clientSession.KeepAlive += Session_KeepAlive;

 if (ServiceResult.IsBad(e.Status))
 {
     var state = sessionReconnectHandler.BeginReconnect(clientSession, basics.ReconnectionInterval, Client_ReconnectComplete);
}`

Normally, it runs every 2 seconds into the SessionKeepAlive function

When the server is manually shut down, the client only enters the Session_KeepAlive function once

The reconnection mechanism is no longer effective

OPCFoundation.NetStandard.Opc.Ua  -  1.5.376.235

marcschier, 2025-07-25:
What is the server you are using, is it the console reference server in this repo?  If not, is the client "trying" to reconnect or is the client session closed?  E.g. due to changing server configuration, etc...