OPCFoundation/UA-.NETStandard #3136: Most GDS tests fail on .net 48 when run locally

marcschier, 2025-07-31:
### Type of issue

- [x] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

[image]

Work on .net 9.0.  

### Expected Behavior

Tests pass consistently on both 48 (while we are supporting it) and .net 9 

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-07-31:
@marcschier Can you paste some traces from the failing tests. On my PC they pass, in the pipelines intermittent failures can also be observed. Again these tests are timing dependent as the server updates the certificate in the background. Do they pass if you increase the wait time?

marcschier, 2025-07-31:
Looks like I am running an old 386, but it is actually pretty beefy machine - 64 GB, 16 cores, workstation, only it is a HD that the repo is on, not SD.  It is however possible that the port is used, or something.

Message: 
OneTimeSetUp: Opc.Ua.ServiceResultException : BadSecureChannelClosed

  Stack Trace: 
<EndAsync>d__12.MoveNext() line 292
--- End of stack trace from previous location where exception was thrown ---
ExceptionDispatchInfo.Throw()
TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
ConfiguredTaskAwaiter.GetResult()
<EndSendRequestAsync>d__9.MoveNext() line 354
--- End of stack trace from previous location where exception was thrown ---
ExceptionDispatchInfo.Throw()
TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
ConfiguredTaskAwaiter.GetResult()
<37 more frames...>
GenericAdapter`1.BlockUntilCompleted()
AsyncToSyncAdapter.Await[TResult](TestExecutionContext context, Func`1 invoke)
SetUpTearDownItem.RunSetUpOrTearDownMethod(TestExecutionContext context, IMethodInfo method)
SetUpTearDownItem.RunSetUp(TestExecutionContext context)
BeforeTestCommand.Execute(TestExecutionContext context)
BeforeTestCommand.Execute(TestExecutionContext context)
BeforeTestCommand.Execute(TestExecutionContext context)
BeforeTestCommand.Execute(TestExecutionContext context)
BeforeTestCommand.Execute(TestExecutionContext context)
CompositeWorkItem.PerformOneTimeSetUp()

marcschier, 2025-07-31:
No, looks like on my machine the key size for a brainpoolP386 is 256 and it crashes.  It is possible I need to update my .net 48 framework, let me try that.

cristipogacean, 2025-08-07:
I cannot replicate on my setup:

[image]