OPCFoundation/UA-.NETStandard #3160: UA Client (SDK) Need to detect changes of own certificate , trust list and crl immediately

charlesaugustineabb, 2025-08-14:
### Type of issue

- [ ] Bug
- [x] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

UA Client (SDK) not  detecting  changes of own certificate , trust list and crl and load it -immediately
for some of use cases currently dependent  on  SecurityTokenLifetime

### Expected Behavior

UA Client (SDK) Need to detect changes of own certificate , trust list and crl and load it -immediately
No Matter what is the value of  SecurityTokenLifetime
During below scenarios 
******************************
1)own certificate is changed due to certificate renewal
2)add or removal of trust list
3)crl is changed or new crl is added or removed

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

cristipogacean, 2025-08-21:
We will have a look on the second half of September.

josefbicik, 2025-09-30:
@cristipogacean do you have any feedback on this topic?